pothosware/SoapyBladeRF #13: Support device loading and programming

karel-, 2018-02-15:
How should the functions in [libbladeRF: Device loading and programming](http://www.nuand.com/libbladeRF-doc/v1.9.0/group___f_n___p_r_o_g.html) be made available from SoapyBladeRF? Especially interested in the device reset functionality.

guruofquality, 2018-02-15:
Theres never been a formal SoapySDR API for that because so many devices do it differently. Automatically on load, command line apps, and obviously APIs -- which in this case is great.

I recommend overloading the `writeSettings(key, value)` call to implement something like 
```
#for example in python it could look like this
device.writeSettings('RESET', 'true');
device.writeSettings('FIRMWARE', open(fw_file).read());
```

karel-, 2018-02-16:
Apparently the firmware files are not well suited to be represented as strings, getting some encoding errors etc. I currently only need the reset function so I am skipping the others.

guruofquality, 2018-02-16:
I may have wrongly assumed that the firmware functions took file contends, given that its a char *, they probably just take the file path, so `device.writeSettings('FIRMWARE', open(fw_file).read());` is really `device.writeSettings('FIRMWARE', fw_file);` FWIW

closed 2018-02-20 by karel-