pothosware/SoapyBladeRF #24: SdrAngel SoapySDR and BladeRF

GkvJeep, 2018-11-24:
Hi .
https://github.com/f4exb/sdrangel/issues/248

guruofquality, 2018-11-24:
Not entirely sure why. I guess all of the settings will get called and their defaults applied? Does this work for you? https://github.com/pothosware/SoapyBladeRF/commit/bfb06500baa772b0bbf925ef176da74e652e97fb

GkvJeep, 2018-11-24:
New problems.
[image]

guruofquality, 2018-11-25:
lol, ok i made that jump to bootloader false too: https://github.com/pothosware/SoapyBladeRF/commit/04f798e6a4c76ab1fb2f3cf19af068f9d473781d

GkvJeep, 2018-11-25:
Hi Josh.
Change resetArg.value to false.

    // Device reset
    SoapySDR::ArgInfo resetArg;
    resetArg.key = "reset";
    resetArg.value = "true";
    resetArg.name = "Reset Device";
    resetArg.description = "Reset the device, causing it to reload its firmware from flash.";

Konstantin.

guruofquality, 2018-12-30:
Should be fixed now, thank!

closed 2018-12-30 by guruofquality