pothosware/SoapyBladeRF #30: setGain no working with bladeRF 2.0

KarlL2, 2019-02-14:
Hi,

I'm not sure this is Soapy related or not. I have a program where I set the gain and then print it, to make sure it was set properly.

It works with a LimeSDR but with the bladeRF 2 the `SoapySDRDevice_setGain` works (ie it returns 0) but then the `SoapySDRDevice_getGain` always returns 60, no matter what I asked. I can even ask for a gain of -500 or +500, I won't get any error, while I would expect a `Provided parameter is out of range`, just like with `setFrequency` if I ask for 10GHz.

guruofquality, 2019-02-14:
The implementation is transparently calling bladerf_get_gain and bladerf_set_gain:

* https://github.com/pothosware/SoapyBladeRF/blob/master/bladeRF_Settings.cpp#L361
* https://github.com/pothosware/SoapyBladeRF/blob/master/bladeRF_Settings.cpp#L381

So it might be a bug in libbladerf. You can put a few prints in there to confirm its libbladerf?

drahosj, 2019-02-14:
The bladeRF, as of the release of the bladeRFmicro, does support AGC. I
wonder if calls to bladerf_set_gain are just silently ignored if gain
control is set to automatic? That would only be the case on RX channels.

KarlL2, 2019-02-15:
Thanks @drahosj, it seems I first need to explicitly disable the gain, before I can set it.

I don't know if it's the expected/usual behavior.

There still seem to be other errors on the libbladerf part, see https://github.com/Nuand/bladeRF/issues/731 .

KarlL2, 2019-06-28:
Actually there is an issue with SoapyBladeRF for the BladeRF2.

In BladeRF [test_gain.c](https://github.com/Nuand/bladeRF/blob/416f048201bfd68bdbbc506720423a85f8fce3d4/host/libraries/libbladeRF_test/test_ctrl/src/test_gain.c#L296), there is

```CPP
/* Enable channel (necessary for RX gain changes to register
* on bladerf2) */
status = bladerf_enable_module(dev, ch, true);
```

I've currently updated the SoapyBladeRF code to this

```CPP
void bladeRF_SoapySDR::setGain(const int direction, const size_t channel, const double value)
{
    // Enable channel (necessary for RX gain changes to register on bladerf2)
    // See bladeRF/host/libraries/libbladeRF_test/test_ctrl/src/test_gain.c
    int ret = bladerf_enable_module(_dev, _toch(direction, channel), true);
    if (ret != 0)
    {
        SoapySDR::logf(SOAPY_SDR_ERROR, "bladerf_enable_module returned %s", _err2str(ret).c_str());
        throw std::runtime_error("setGain() " + _err2str(ret));
    }

    ret = bladerf_set_gain(_dev, _toch(direction, channel), bladerf_gain(std::round(value)));
    if (ret != 0)
    {
        SoapySDR::logf(SOAPY_SDR_ERROR, "bladerf_set_gain(%f) returned %s", value, _err2str(ret).c_str());
        throw std::runtime_error("setGain() " + _err2str(ret));
    }
}
```

I'm not sure if it's possible to know if the channel is already enabled to avoid redoing it, nor if it causes issues if it's already opened.

guruofquality, 2019-06-28:
It seems like we should be able to call bladerf_enable_module in the constructor then. Can you try removing it from setup/closeStream, and move them to the constructor instead?

KarlL2, 2019-06-28:
It seems to be working (I haven't done extensive tests). I removed the `bladerf_enable_module` calls from `setupStream` and `closeStream` and created a simple function to enable or disable all channels in both directions, that I call in the constructor and destructor:

```CPP
void bladeRF_SoapySDR::_enable_or_disable_module(const bool enable, const bool throwOnError) {
  std::vector<int> directions;
  directions.push_back(SOAPY_SDR_RX);
  directions.push_back(SOAPY_SDR_TX);

  std::vector<size_t> channels;
  channels.push_back(0);
  if (_isBladeRF2)
    channels.push_back(1);

  for (const auto dir : directions) {
    for (const auto ch : channels)
    {
      const int ret = bladerf_enable_module(_dev, _toch(dir, ch), enable);
      if (ret != 0)
      {
        SoapySDR::logf(SOAPY_SDR_ERROR, "bladerf_enable_module(%s) returned %s", enable ? "true" : "false", _err2str(ret).c_str());
        if (throwOnError)
          throw std::runtime_error("bladerf_enable_module() " + _err2str(ret));
      }
    }
  }

  directions.clear();
  channels.clear();
}
```

closed 2019-07-01 by guruofquality