pothosware/SoapyBladeRF #38: BladeRf 2.0 multiple input - 2 different frequences on 2 different inputs

mateusz-spychala, 2020-10-26:
Hello,

I use Bladerf 2.0 A4 with soapysdr and gr-soapy.

It want to set 2 different frequencies on 2 input channels, but it seems that
both channels are set to same frequency.

Is it possible?

guruofquality, 2020-10-28:
I think there is only one local oscillator per direction on this hardware, so you only get one frequency for tx, one for rx

mateusz-spychala, 2020-10-29:
I have checked in documentation and it is a bit misleading. 
According to block schematic it should be possible:
[Block diagram](https://ez.analog.com/cfs-file/__key/communityserver-discussions-components-files/441/ad9412711885edbee3abd6547b0ac3e4.png)
It seems when you switch RX part to TX oscillator you also have to use TX-MON input (that do not have LNA). 
Summarizing in general case it is possible (external LNA required), on BladeRf hardware it is not possible.

closed 2020-10-29 by mateusz-spychala