pothosware/SoapyBladeRF #39: Issue with Bladerf2 + soapy (+ osmosdr)

ikarus23, 2020-11-13:
Hi. I have an up to date Arch Linux, an Bladerf 2.0 micro and osmosdr. When I use `osmocom_fft` with the native bladerf driver, everything works fine. When I use it with the soapy driver, the graph jitters and I get overflows.
```
osmocom_fft -a bladerf -g 50 -s 4e6 -f 433e6   # works great
osmocom_fft -a soapy -g 50 -s 4e6 -f 433e6     # makes issues
```
Video of the issue: http://tests.icaria.de/bladerf-soapy.mkv

Does anybody have the same issues? Any idea what might cause this?

alphafox02, 2021-02-05:
Same here, I copied and pasted your commands and I can see a noticeable difference between the two. The soapy attempt jitters and has overflows.

guruofquality, 2021-02-08:
* https://github.com/pothosware/SoapyBladeRF/blob/master/bladeRF_Streaming.cpp
* https://github.com/osmocom/gr-osmosdr/blob/master/lib/bladerf/bladerf_source_c.cc

Might be worth printing the parameters passed to bladerf_sync_config and bladerf_sync_rx. Maybe one of the buffer sizes are way off.