pothosware/SoapyBladeRF #44: 2-Channel Receive Error with libbladerf 2021.10

AlexFWulff, 2021-11-09:
On libbladerf 2021.10 with a bladeRF Micro 2.0 xA4 the following Gist will not correctly receive samples:
[https://gist.github.com/AlexFWulff/9c821eb999e9da86404a163fb35b2656](https://gist.github.com/AlexFWulff/9c821eb999e9da86404a163fb35b2656)

I consistently get errors from libbladerf and cannot receive samples properly. I downgraded libbladerf to 2019.07 (and also downgraded the firmware of my SDR and FPGA), and now this code works with no issues. It appears that somewhere between these two releases something was introduced that breaks 2-channel reception with SoapySDR. On either release I can receive from just one channel with no issues.

guruofquality, 2021-11-14:
Are there any samples being received, or do they just happen to be zeros, or randomly zeroed out?

I wonder if its a weird buffer size difference that no longer works for some reason. Can you print whats passed to bladerf_sync_config:

* soapy blade rf https://github.com/pothosware/SoapyBladeRF/blob/master/bladeRF_Streaming.cpp#L132
* osmo https://github.com/osmocom/gr-osmosdr/blob/master/lib/bladerf/bladerf_source_c.cc#L224

Futher do you ever see the code get to bladerf_sync_rx?  https://github.com/pothosware/SoapyBladeRF/blob/master/bladeRF_Streaming.cpp#L324

AlexFWulff, 2021-11-21:
Some samples are received _some of the time_, but never as many as request and oftentimes the receive call will fail entirely. I posted more details on the Nuand forum here with the exact lines in which things are going wrong: https://nuand.com/forums/viewtopic.php?f=5&t=12874 - let me know if you still need more information!