pothosware/SoapyBladeRF #45: SoapyBladeRF usb timeout errors

akoun, 2022-01-05:
Hello
I try to test BladeRF under Pothos (built from source) and I get strange errors when using the bladerf soapy sink

[22:22:54.214000] SoapySDR: bladerf_open_with_devinfo()
[22:22:54.394000] SoapySDR: bladerf_get_serial() = 1d62a870f59e65fdd231ced37787b03a
[22:22:54.401000] SoapySDR: setSampleRate(Rx, 0, 4.000000 MHz), actual = 4.000000 MHz
[22:22:54.409000] SoapySDR: setSampleRate(Tx, 0, 4.000000 MHz), actual = 4.000000 MHz
[22:22:54.417000] SoapySDR: setSampleRate(Rx, 0, 4.800000 MHz), actual = 4.800000 MHz
[22:22:54.421000] ::1: [INFO @ fpga_common/src/lms.c:2258] Clamping frequency to 237500000Hz
[22:23:04.972000] ::1: [INFO @ fpga_common/src/lms.c:2258] Clamping frequency to 237500000Hz
[22:23:05.017000] : 0
[22:23:05.248000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.249000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.250000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.348000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.349000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.350000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.447000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.448000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.449000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.547000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
.....

the flowgraph is stuck and does not do anything. In gnuradio-companion the BladeRF source behaves similarly but curiously when  gr-osmosdr sink everything works nicely.

If somebody can help I'd greatly appreciate because I am completely stuck with this.
best regards
aK

akoun, 2022-01-06:
Hello, 

I was able to resolve my problem of usb transfer time-outs. Here is the workaround:
I downgraded the FPGA image to:
2019-07-31 - v0.11.0
and the FX3 firmware to:
2018-12-16 - v2.3.2

and then the errors disappeared...

I am under Debian bookworm with libbladerf version 0.2021.10-2 (I was planning to downgrade this as well to 0.2019.07-7 but finally it was not necessary).

I hope this can be helpful because I chased my tail around for a while with this one.

cheers,
aK

davwheat, 2023-03-19:
Unfortunately this isn't a solution for the xA5 since it was released after those versions.