pothosware/SoapyBladeRF #48: Update dual streaming patch not building

alphafox02, 2022-05-09:
It can of course be something on my end, but while trying try to build the latest source with latest comment I get the following,

```
/opt/build/SoapyBladeRF/bladeRF_Streaming.cpp:320:10: error: ‘memset’ is not a member of ‘std’
  320 |     std::memset(&md, 0, sizeof(md));
      |          ^~~~~~
/opt/build/SoapyBladeRF/bladeRF_Streaming.cpp: In member function ‘virtual int bladeRF_SoapySDR::writeStream(SoapySDR::Stream*, const void* const*, size_t, int&, long long int, long int)’:
/opt/build/SoapyBladeRF/bladeRF_Streaming.cpp:420:10: error: ‘memset’ is not a member of ‘std’
  420 |     std::memset(&md, 0, sizeof(md));
      |          ^~~~~~
make[2]: *** [CMakeFiles/bladeRFSupport.dir/build.make:89: CMakeFiles/bladeRFSupport.dir/bladeRF_Streaming.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:104: CMakeFiles/bladeRFSupport.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

However, if I roll one commit back it builds just fine.

alphafox02, 2022-05-09:
Confirmed two commits ago will build 
70505a5cdf8c9deabc4af3eb3384aa82a7b6f021

This is on 20.04 w/ lib version 2.4.1-git-5a146b2a

guruofquality, 2022-05-09:
try to pull, i added a missing include d0bae0c61a7f89ee000c727197fb06746893b692

alphafox02, 2022-05-10:
After pulling there's no issues building now. Thank you.

closed 2022-05-10 by alphafox02