pothosware/SoapyBladeRF #53: ‘SOAPY_SDR_USER_FLAG1’ was not declared in this scope

alphafox02, 2022-06-05:
Just noticed this will building from latest source.

```
/opt/build/SoapyBladeRF/bladeRF_Streaming.cpp: In member function ‘virtual int bladeRF_SoapySDR::readStream(SoapySDR::Stream*, void* const*, size_t, int&, long long int&, long int)’:
/opt/build/SoapyBladeRF/bladeRF_Streaming.cpp:395:71: error: ‘SOAPY_SDR_USER_FLAG0’ was not declared in this scope; did you mean ‘SOAPY_SDR_OVERFLOW’?
  395 |     if ((md.status & BLADERF_META_FLAG_RX_HW_MINIEXP1) != 0) flags |= SOAPY_SDR_USER_FLAG0;
      |                                                                       ^~~~~~~~~~~~~~~~~~~~
      |                                                                       SOAPY_SDR_OVERFLOW
```

guruofquality, 2022-06-05:
of course that only took less than a day :laughing:  https://github.com/pothosware/SoapyBladeRF/pull/49#issuecomment-1146608299

Can you pull master again, i added a compile time check for the new flags

alphafox02, 2022-06-05:
Haha, this was just random timing, I promise! Pulled master and rebuilt without issue. Thank you.

closed 2022-06-05 by alphafox02