pothosware/SoapyBladeRF #54: Consider supplying the quick_tune parameter to bladerf_schedule_retune() when the XB-200 is enabled.

ratzrattillo, 2022-06-28:
When trying to tune to frequencies below 300Mhz using BladeRF 1.0 with the XB-200 extension board enabled, the following error message is shown usuallly:
`[ERROR @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:2398] Consider supplying the quick_tune parameter to bladerf_schedule_retune() when the XB-200 is enabled.`

Would it be possible to automatically supply this quick_tune parameter in SoapySDR, so that the error disappears?

guruofquality, 2022-06-28:
@ratzrattillo Can you make a merge request for the change? There are a number of cases that setFrequency handles for v1 vs v1 with and without quick tune. I think with the right getters it could be automatic for the  XB-200. Take a look at the setFrequency code here: https://github.com/pothosware/SoapyBladeRF/blob/master/bladeRF_Settings.cpp#L429

ratzrattillo, 2022-10-03:
I tried to experiment and figure out how to do it. I compiled both SoapySDR and the SoapyBladeRF module from source. However, my BladeRF is not even detected when compiling the latest master branch for both projects.
Using: `SoapySDRUtil.exe --find`
```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[ERROR] SoapySDR::loadModule(C:\Program Files (x86)\SoapySDR/lib/SoapySDR/modules0.8-2/bladeRFSupport.dll)
  bladerf failed ABI check: Library ABI=0.8-2, Module ABI=0.8
No devices found!
```
Is there a new release planned soon, so the ABIs will amtch again?
Can the ABI version be adjusted by the developers?

ratzrattillo, 2022-10-15:
While still being interested in using the bladerf after compiling from source and solving the ABI issue, i opened an issue in the main bladerf repository from NUAND, referencing this error, suggesting also a way to mitigate it:

https://github.com/Nuand/bladeRF/issues/882

Just FYI @guruofquality

guruofquality, 2022-10-16:
> referencing this error, suggesting also a way to mitigate it: 

cool

> Is there a new release planned soon, so the ABIs will amtch again?
Can the ABI version be adjusted by the developers?

The error just means the version of the headers the bladeRFSupport.dll was compiled against dont match the version of the runtime library SoapySDR.dll So you have some kind of dev environment or %PATH% issue.

The print of mismatch is sanity check to avoid what would otherwise be unexplained crashiness