pothosware/SoapyBladeRF #55: Add new release tag

Heshyo, 2022-11-28:
The available [package on Ubuntu is 0.4.1](https://packages.ubuntu.com/jammy/soapysdr-module-bladerf), which dates from Jan 2019.

Could a new tag be added? Also should the maintainers be contacted to update the package?

KarlL2, 2022-11-28:
Good idea, but please merge https://github.com/pothosware/SoapyBladeRF/pull/56 first.

Heshyo, 2024-12-20:
@guruofquality Could a new tag be added and a new package released on debian/ubuntu? There's been a few useful modifications since the last package.

guruofquality, 2024-12-21:
> @guruofquality Could a new tag be added and a new package released on debian/ubuntu? There's been a few useful modifications since the last package.

Good idea. I am updating the changelog: https://github.com/pothosware/SoapyBladeRF/pull/63

guruofquality, 2024-12-22:
https://github.com/pothosware/SoapyBladeRF/tree/soapy-bladerf-0.4.2

closed 2024-12-22 by guruofquality

KarlL2, 2025-01-06:
Thanks Josh. Maybe SoapySDR also needs to have a new release tag, as some new constants were introduced: https://github.com/pothosware/SoapySDR/pull/353 (unless you've handled the case in SoapyBladeRF when they don't exist https://github.com/pothosware/SoapyBladeRF/pull/49#issuecomment-1146608299 ).